Bouni/kicad-jlcpcb-tools #635: Part selector shows limited results

adamjsmithnz, 2025-07-22:
**Describe the bug**
The JLC parts search function is no longer showing results for a search query especially for basic parts. I successfully used the selector last week and it remained working until I updated the parts database earlier today.
JLC seems to have done an update to their parts library recently including renaming Preferred Extended Parts to Promotional Extended. I wonder if this is a coincidence or somehow related.

**To Reproduce**
Steps to reproduce the behavior:
1. Open JLCPCB Tools and download a new copy of the parts database
2. Search for any part, eg 100n 0603 with filter for Basic parts only. I get 0 results
3. If you then search for the corresponding LCSC part number, eg C14663, the params field is just '0603' and the description just 'not ROHS'

**Expected behavior**
The list of search results matches those on JLC parts selector website.

# KiCad Version

```
Application: KiCad PCB Editor x64 on x64

Version: 9.0.2, release build

Libraries:
	wxWidgets 3.2.6
	FreeType 2.13.3
	HarfBuzz 10.2.0
	FontConfig 2.15.0
	libcurl/8.11.1-DEV Schannel zlib/1.3.1

Platform: Windows 11 (build 22631), 64-bit edition, 64 bit, Little endian, wxMSW
OpenGL: NVIDIA Corporation, NVIDIA GeForce RTX 4060/PCIe/SSE2, 4.6.0 NVIDIA 566.24

Build Info:
	Date: May  7 2025 02:29:49
	wxWidgets: 3.2.6 (wchar_t,wx containers)
	Boost: 1.86.0
	OCC: 7.8.1
	Curl: 8.11.1-DEV
	ngspice: 44
	Compiler: Visual C++ 1942 without C++ ABI
	KICAD_IPC_API=ON

Locale: 
	Lang: en_US
	Enc: UTF-8
	Num: 1,234.5
	Encoded кΩ丈: D0BACEA9E4B888 (sys), D0BACEA9E4B888 (utf8)
```
JLCPCB Tools 2025.04.02

nippoo, 2025-07-22:
This is also the issue I found. The problem seems to be that the "Description" column is just "not ROHS" for everything, even if I search for the part by name. So it only matches ones where the MFR number happens to match (which is very few parts)

[image]

[image]

chmorgan, 2025-07-22:
> <https://github.com/user-attachments/assets/9d7a0675-389e-49c6-b90f-30504f357e05> Screenshot.2025-07-22.at.12.59.59.png

Can you make a separate issue for the not rohs? I can look at that shortly
as well.

chmorgan, 2025-07-23:
> Can you make a separate issue for the not rohs? I can look at that shortly
> as well.
> […](#)

@nippoo created #637, its due to an issue with the upstream parts library

nippoo, 2025-07-23:
Hi @chmorgan, this is the issue for the 'not ROHS', I think, the new issue you've just created is a duplicate of this one?

chmorgan, 2025-07-23:
@nippoo it could be considered a duplicate but I wanted to capture the root cause clearly. We can close them both out when we wrap things up.

LeandroMarceddu, 2025-08-03:
Bump. I hit the update catalog button and now nothing is showing up either, all due to No ROHS