Bouni/kicad-jlcpcb-tools #637: parts are missing descriptions - this is due to upstream (https://github.com/yaqwsx/jlcparts/issues/151)

chmorgan, 2025-07-23:
https://github.com/yaqwsx/jlcparts/issues/151

c-ewing, 2025-07-24:
Heyo, I ran into this issue but found that a fork of the original was still able to find components: https://github.com/dougy83/jlcparts. I don't know if this will help at all. I will take a better look at the fork and see if i can track down a specific change.

redfast00, 2025-07-25:
@chmorgan maybe it could be useful to switch to a more maintained fork https://github.com/dougy83/jlcparts

chmorgan, 2025-08-14:
@Bouni as an update here, I was denied a lcsc api key twice. We may be able to get a jlcpcb api key but it's unclear if this is sufficient to rebuild the database. Thoughts on how to proceed? Without the database we are kind of in trouble with the plugin.

Bouni, 2025-08-15:
Hey I got sort of a special access from them 1-2 years ago but that was not API access. It allowed me to download the database manually from the web UI. I tried to script that but the captcha made it impossible to do so.

I can try to convince them to give me access. 

I'll let you know what they say.

@chmorgan where did you try to get access? Via the web form?

chmorgan, 2025-08-15:
> Quote from @yaqwsx <https://github.com/yaqwsx> jlcparts Readme:
>
> Travis CI download XLS spreadsheet from the JLC PCB page, a Python script
> process it and it generates per-category JSON file with components.
>
> So I guess he's not using the API. Maybe they no longer offer the xls
> file!?

The GitHub workflow has both jlcpcb and lcsc keys, passed as secrets to the
build.

I believe the list is downloaded and then parts are checked and only the
first 1000 are updated, by pulling from the lcsc entries from the api.
(This is the part I can say for sure I remember seeing yet.)

yaqwsx, 2025-08-16:
We fetch a list of LCSC codes available on JLCPCB via their API. This happens every time. Then, we use the heavy rate-limited API of LCSC to fetch attributes of components (e.g., resistance, number of pins, inductance, Rdson and similar). These values are pretty much stable, so we cache them. In order not to go out of sync over long time spans, we refresh these values for the n-oldest components. Caching is a mechanism to work around the rate-limited API of LCSC.

I am sorry for my unresponsiveness. Either JLCPCB or LSCL changed their API without notice, and I am on vacation with limited access to developer tools.

yaqwsx, 2025-08-16:
Hello guys, can you please precisely point me to the problem? I managed to download the newest SQLite DB provided by JLCParts, and I can see that practically all components have descriptions. Only 1000 components out of the 4 million components have no description.

yaqwsx, 2025-08-17:
@chmorgan: Of course it is used: https://github.com/yaqwsx/jlcparts/blob/5fa91e1562a69d0abdd0246d11d20f07c3cb28ca/jlcparts/ui.py#L58

pggood, 2025-09-04:
Ive done an update that gets rid of the parts database and gets the info from https://jlcpcb.com/parts/ its quite a bit slower than using the sqlite database but the information is the current including inventory its based on an earlier version as Im stillusing kicad 6.0 

[KiCAD-PCM-2021.12.1.zip](https://github.com/user-attachments/files/22162472/KiCAD-PCM-2021.12.1.zip)

pggood, 2025-09-05:
as above with 

[KiCAD-PCM-2021.12.1.zip](https://github.com/user-attachments/files/22164674/KiCAD-PCM-2021.12.1.zip)

better description

pggood, 2025-09-05:
> @pggood<https://github.com/pggood> Could share how you download the component list? If you have a reliable source of data, I can fix jlcparts.

Hi Jan

Ive replaced the sqlite search in library.py with the JLCPCB web api the two files Ive also changed mainwindow.py to remove the download button and partselector.py to add right click and also the updatelist to cope with the list format returned from jlcpcb

Kind regards
Peter

pggood, 2025-09-07:
Checks to see if the parts on the PCB have been updated 

[KiCAD-PCM-2021.12.1.zip](https://github.com/user-attachments/files/22199033/KiCAD-PCM-2021.12.1.zip)

pggood, 2025-09-08:
> @pggood<https://github.com/pggood> : Do you have a code you could share? I am not able to taky any action based on your description. The JLCParts need a list of all components provided by JLC PCB. How do you construct yours?

Hi Jan
I made to the JLCPCB Tools plugin. You mentioned you're unable to act on my description alone and asked about how I construct the list of components without the downloaded SQLite database. I'll explain the approach below and share the relevant code snippets.
High-Level Overview
In the original plugin, part searches relied on a pre-downloaded SQLite database (e.g., parts.db) containing a static snapshot of JLCPCB's component library. This had limitations like needing periodic updates, large file sizes, and potential staleness (e.g., out-of-date stock or pricing).
I replaced this with dynamic web-based lookups using JLCPCB's public APIs (specifically their component search endpoints). This:

Fetches results in real-time from JLCPCB's servers.
Avoids downloading/maintaining a large local database.
Handles pagination, filtering (e.g., by keyword, manufacturer, package, category, stock), and sorting dynamically.
Falls back to a "suggest" API if the main search fails.
Respects API rate limits with delays and retries (e.g., 0.5-2s between requests).
Mimics browser behavior (headers, cookies, XSRF tokens) to avoid blocks, as the APIs are undocumented and scraped from their web app.

The core logic lives in a new JLCPCBSearch class in library.py. The Library class's search() method now calls this instead of querying SQLite. Results are parsed into a list of dicts (similar to the old DB rows) and returned for display in the PartSelectorDialog.
This keeps the user-facing behavior the same (e.g., search fields in PartSelectorDialog still work), but the backend is now API-driven. It requires an internet connection, but that's a reasonable trade-off for up-to-date data.
How the Component List is Constructed

No Local List/Database: Instead of building/maintaining a full list of all JLCPCB components locally, we query their search API on-demand. For a given search (e.g., keyword="10k 0603", manufacturer="Vishay"), we:

Initialize a session by visiting their search page (to grab cookies/XSRF tokens).
Send a POST to their main component search API (/api/overseas-pcb-order/v1/shoppingCart/smtGood/selectSmtComponentList/v2).
Paginate results (up to 6 pages of 25 results each = max 150 results, to avoid overload).
Filter client-side (e.g., by stock, library type like Basic/Extended).
Parse JSON responses into a list of component dicts (e.g., with keys like 'JLCPCB Part #', 'Stock', 'Price').

Fallback: If the main API fails (e.g., due to rate limits or 403 errors), we try their "suggest" API (/api/overseas-pcb-order/v1/shoppingCart/smtGood/suggestComponent).
Stock Handling: Stock is fetched dynamically per part. In mainwindow.py, there's an "Update Stock" button that queries individual part details via another API (/shoppingCart/smtGood/getComponentDetail).
Error Handling: Retries on 403s, logs errors, and gracefully handles no-results cases.
Limitations: API is unofficial, so it could break if JLCPCB changes their site. We limit to 150 results to stay under radar. No full catalog download—just targeted searches.

This constructs the list lazily based on user input, rather than pre-loading everything.
Key Code Snippets
Here are the main changes from library.py (the rest of the file handles rotations, which I didn't touch). I've omitted some logging/debug code for brevity, but the full file is in the attached/linked repo [if you have one, link it here].
1. The JLCPCBSearch Class (Handles API Requests)
This is the new core for web queries. It manages sessions, headers, compression (gzip/deflate), and retries.

import json
import time
import urllib.request
import urllib.parse
import ssl
import gzip
import zlib
import http.cookiejar
from html.parser import HTMLParser

class XSRFParser(HTMLParser):
    def __init__(self):
        super().__init__()
        self.xsrf_token = None

    def handle_starttag(self, tag, attrs):
        attrs_dict = dict(attrs)
        if tag == 'meta' and ('name', 'csrf-token') in attrs:
            self.xsrf_token = attrs_dict.get('content')
        elif tag == 'meta' and ('name', '_token') in attrs:
            self.xsrf_token = attrs_dict.get('content')
        elif tag == 'input' and ('name', '_token') in attrs:
            self.xsrf_token = attrs_dict.get('value')

class JLCPCBSearch:
    def __init__(self, debug=False):
        self.base_url = "https://jlcpcb.com"
        self.base_headers = {
            "User-Agent": "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/139.0.0.0 Safari/537.36",
            "Accept": "application/json, text/plain, */*",
            "Accept-Language": "en-US,en;q=0.9",
            "Accept-Encoding": "gzip, deflate",
            "Connection": "keep-alive",
            "Cache-Control": "max-age=0",
[... 76 lines cut ...]
                str(item.get('MFR.Part #', 'N/A')),
                str(item.get('Package', 'N/A')),
                str(item.get('Library Type', 'N/A')),
                str(item.get('Manufacturer', 'N/A')),
                str(item.get('Description', 'N/A')),
                str(item.get('Price', 'N/A')),
                str(item.get('Stock', 0)),
            ]
            output.append(row)
        return output[:150]  # Cap results

3. Integration with PartSelectorDialog
No major changes here—the populate_part_list() method in partselector.py still takes the list from library.search() and displays it in the DataViewListCtrl. It works seamlessly with the new dynamic results.
[... cut ...]
  *   Works Offline? No, requires internet. Falls back gracefully if no connection.
  *   Rate Limits: Delays between pages; retries on 403s.
  *   Example Query: Keyword="10k 0603", Manufacturer="Vishay", Package="0603", Category="Resistor", In-Stock Only.
  *   Performance: For typical searches, 1-2s per page. Total <10s for 150 results.
  *   Full Code: If you'd like the complete modified files (library.py, etc.), I can share a GitHub repo/fork or zip file—just let me know.

Ive attached the complete code with the changes Im still using kicad 6.0

Regards
Peter Goodrcke

pggood, 2025-09-18:
If you are using kicad 9 there are a couple of changes that had errors 

[KiCAD-PCM-2021.12.1_9.zip](https://github.com/user-attachments/files/22413949/KiCAD-PCM-2021.12.1_9.zip)

wburchard, 2025-09-18:
Would there be any interest in a script to scrape
https://jlcpcb.com/parts/basic_parts?  Curiosity pulled me in and I figured
out how to do it.

z2amiller, 2026-04-10:
Fixed by: #670